stm32-rs/stm32wlxx-hal #291: Include Example of DMA usage

justacec, 2022-03-09:
I am working to use this crate to drive my LoRa-E5 and right now just trying to get UART comms with DMA running.  After a review of the crate, it seems as though all of the low-level DMA bit flipping is happening in the trait impl's.  So, to enable DMA transfers, one just needs to enable the tx/rx with a DMA tuple.  But, I am confused on the "correct/idiomatic" way to use these trait impl's.  Can you provide a simple UART with DMA example, something like a simple echo server.

Specific questions I would look to answer with this would be, 

   1. Do I do `enable_rx` and `enable_tx` at UART instantiation or just immediately prior to use the Write or Read functions?
   2. The way the code is written, it does not seem to be able to pass a vector of u8 to a write function in a non-blocking way.  Historically, I would set the men address of the DMA channel, set the number of transfers, enable, and then walk away and do other things letting the DMA irq tell me when it is done.  This does not currently seem possible. 
  
I think an example will help me see how this was intended to be done.

newAM, 2022-03-09:
1. Enable RX / enable TX can be called once as part of setup.  DMAR/DMAT just connect internal signals, but do nothing if the DMAMUX is not listening to the signal.
2. You are correct, this is not currently possible in the existing implementation.

There is an example usage with two UARTs talking to each other in the unit tests: https://github.com/stm32-rs/stm32wlxx-hal/blob/main/testsuite/src/uart.rs

I try to put most things that would normally be examples into unit tests because they get checked in CI.  If that needs more expansion / clarification let me know!  I have not had much feedback on the code 😅

justacec, 2022-05-28:
Just revisiting this to say that I have this working with DMA now.  (using the blocking approach). Might be fine to go ahead and close this issue.

newAM, 2022-05-28:
Alright, feel free to re-open :+1:

closed 2022-05-28 by newAM