stm32-rs/stm32wlxx-hal #314: can not build success

CamelShoko, 2022-05-19:
Hi, i clone this code, but it can not build successful.

```
$ DEFMT_LOG=trace cargo run-ex gpio-blink
error: failed to load manifest for workspace member `/Users/zhangxiang/embedded-worksapce/DLB/Code/stm32wlxx-hal/lora-e5-bsp`

Caused by:
  failed to parse manifest at `/Users/zhangxiang/embedded-worksapce/DLB/Code/stm32wlxx-hal/lora-e5-bsp/Cargo.toml`

Caused by:
  namespaced features with the `dep:` prefix are only allowed on the nightly channel and requires the `-Z namespaced-features` flag on the command-line
```

newAM, 2022-05-19:
Try update your rustc version, this only supports rust 1.60+ at the moment.  If you are using `rustup` this is just `rustup update`

CamelShoko, 2022-05-19:
when i do rustup update
then use 
$ cd examples
$ cargo build 

`LLVM ERROR: Global variable '__INTERRUPTS' has an invalid section specifier '.vector_table.interrupts': mach-o section specifier requires a segment and section separated by a comma.
error: could not compile `stm32wl`
warning: build failed, waiting for other jobs to finish...
error: build failed`

CamelShoko, 2022-05-19:
`DEFMT_LOG=trace cargo run-ex gpio-blink`

```
    Finished dev [optimized + debuginfo] target(s) in 0.04s
     Running `probe-run --chip STM32WLE5JCIx --connect-under-reset /Users/zhangxiang/embedded-worksapce/DLB/Code/stm32wlxx-hal/target/thumbv7em-none-eabi/debug/examples/gpio-blink`
(HOST) INFO  flashing program (11 pages / 11.00 KiB)
(HOST) INFO  success!
────────────────────────────────────────────────────────────────────────────────
INFO  Starting blinky
└─ gpio_blink::__cortex_m_rt_main @ examples/gpio-blink.rs:33

```

the 'rustup update' useful, thank you.

closed 2022-05-19 by newAM