stm32-rs/stm32wlxx-hal #319: How to Establish a Continuous LPTIM Peripheral

justacec, 2022-05-28:
I would like to use an LPTIM peripheral to track time.  To do so I would like it to be in continuous mode so it just fires off the event and then reloads and keeps going.  Unfortunately, the `start` function implementation for the LPTIM peripherals seems to set the mode to Single.  How would I go about setting this up for continuous?

newAM, 2022-05-28:
Continuous timers are part of the `PwmPin` traits:
https://github.com/stm32-rs/stm32wlxx-hal/blob/a9d4b5a77d7d4fc6d5da3402328196d980d4b680/hal/src/lptim/mod.rs#L767

There's an example here: https://docs.rs/stm32wlxx-hal/latest/stm32wlxx_hal/lptim/trait.LpTim.html#example-5

The `new_output_pin` method is optional, that's just for PWM.

justacec, 2022-05-28:
Ahh.  Got it.  I should have seen that.  I will take a look there and see if I can get everything to work.

Thanks for being so helpful and responsive.  :)

newAM, 2022-05-28:
No problem!  It is misleading being called `PwmPin` when the pin isn't even required :sweat_smile: