stm32-rs/stm32wlxx-hal #339: LPUART type mismatch & some other issues about UART peripheral implementation

huming2207, 2022-09-12:
Hi all,

Sorry for this stupid question, I'm new to Embedded Rust. I'm trying to use this HAL crate with RTIC together to implement a UART-LoRa modem. But so far I just couldn't figure out this issue when I try to enable `RXNEIE` (receive data register not empty interrupt):

[image]

Here's the code: https://github.com/huming2207/lowu-basestation/blob/cdf87dbade6c983cbe4c77556b336dfc17320af9/src/bin/app.rs

I've also tried replacing the `uart_p` to `dp.LPUART` but still no luck, and Rust compiler complains the exact same error when compiling so it's not a language server or VSCode plugin's false positive. Anyway, may I ask if is there any suggestion on how to get `RXNEIE` to be set?

Meanwhile I've also got another issue with `defmt-serial` earlier in July: https://github.com/gauteh/defmt-serial/issues/1

Basically, the author tested his implementation works fine on some other microcontrollers, but it didn't work at all on my RAK3172 (STM32WLE5JC) development kit. I'm not sure if this issue is related to the USART/LPUART implementation in this crate, but I've brought it here just in case.

Regards,
Jackson

newAM, 2022-09-13:
Ouch, that is an unfortunately difficult error message :(

What it is saying is that the closure `|_, w| { ... }` expects a different return type.

```rust
uart_p.cr1.modify(|_, w| {
    w.rxneie().enabled();
});
```

The semicolon after `enabled()` makes it look like this to the compiler:

```rust
uart_p.cr1.modify(|_, w| {
    w.rxneie().enabled();
    return ();
});
```

Without the semicolon it compiles:

```rust
uart_p.cr1.modify(|_, w| w.rxneie().enabled());
```

---

For the other issue I'm unfamiliar with `defmt-serial`, will take a closer look when I have more time this weekend :)

huming2207, 2022-09-13:
Ah oops, my bad! 😅  Thanks for your info!

I will have a try again later tonight.

huming2207, 2022-09-13:
I can now confirm it's working. Sorry again, I completely misread the error message earlier.

closed 2022-09-13 by huming2207

newAM, 2022-09-13:
No worries, I've hit that one myself, embedded rust can be difficult at times, happy to help out!