stm32-rs/stm32wlxx-hal #346: how to access i2c

MrSoUndso, 2022-11-21:
Hi @newAM!

Thanks for your work on this library, it's quite helpful!

Sadly, I'm struggling to get I2C to work. I'm quite new to (embedded) rust, so this might just be a small error in my understanding.

Here is some example code

```
use cortex_m_rt::entry;

use stm32wlxx_hal::{
    pac::{self}, gpio::{PortA}, i2c, embedded_hal::prelude::_embedded_hal_blocking_i2c_Write
};

#[entry]
fn main() -> ! {
    let mut dp: pac::Peripherals = pac::Peripherals::take().unwrap();
    let cp: pac::CorePeripherals = pac::CorePeripherals::take().unwrap();

    let gpioa: PortA = PortA::split(dp.GPIOA, &mut dp.RCC);

    let scl = gpioa.a11;
    let sda = gpioa.a12;

    let i2c = cortex_m::interrupt::free(|cs| {
            i2c::I2c2::new(dp.I2C2, (scl, scl), 1500, &mut dp.RCC, true, cs)
        });
    
    i2c.write(0x56, &[0]); //example data

    loop {
        hprintln!("Running!").unwrap();
    }
}

```

This fails on `i2c::I2c2::new` with `the trait `gpio::sealed::I2c2Scl` is not implemented for 'A11'`.

How can I resolve this? 
Thanks in advance for any help you can provide me :)

MrSoUndso, 2022-11-21:
Nvm, I just found out that i had scl and sda backwards.

I found the solution in the unit tests, thanks to your comment on issue #291. 

I might be helpful to add a comment like this to the readme, so other users find these examples faster.

newAM, 2022-11-21:
Yeah I need to improve the README with respect to the discoverability of the examples.

I'll leave this issue open until that's fixed 🙂

PR #347 (2022-11-26): README: emphasize on-target tests
Closes #346
closed 2022-11-26 by newAM