stm32-rs/stm32wlxx-hal #368: compile

tcpipchip, 2023-09-08:
Hi,  i want to test your gpio-blink.rs in my module LSM110A (W55JC)...

I am doing on examples folder

$~/stm32wlxx-hal
cd examples
cargo build

[image]

How i create the hex to burn in my Module ?

I dont want to use the cargo-run

Thanks

newAM, 2023-09-09:
Building creates an ELF file (under the `target` directory).  `objcopy` can be used to change the format of that file.

tcpipchip, 2023-09-09:
hi, but wich name ?

tcpipchip, 2023-09-10:
well, solved...
I did a blog too
https://lom204-cli-wisol.blogspot.com/2023/09/lsm110a-blink-led-com-rust.html

newAM, 2023-09-10:
Glad you got it working!  Sorry I couldn't provide the path earlier, I'm away from my computer for another week.

I'll close to issue now, but feel free to reopen!

closed 2023-09-10 by newAM