stm32-rs/stm32wlxx-hal #371: Lora P2P communication

kolegs, 2023-10-29:
Hi,

I am a rust embedded newbie and I am trying to configure Lora P2P communication between two NUCLEO-WL55JC boards. But I got a problem with even sending any data. When I check `radio.irq_status()` after calling `set_tx` I get `ExecutionFailure`, which means that the command was accepted but the data couldn't be sent. Is there any example of Lora configuration available for a reference?

Below configuration code:
```
    let (ctrl1, ctrl2, ctrl3) = cortex_m::interrupt::free(|cs| {
        (
            Output::default(gpioc.c4, cs),
            Output::default(gpioc.c5, cs),
            Output::default(gpioc.c3, cs),
        )
    });

    let mut radio_switch = radio_switch::RadioSwitch::new(ctrl1, ctrl2, ctrl3);

    const TX_BUF_OFFSET: u8 = 128;
    const RX_BUF_OFFSET: u8 = 0;

    const DATA_LEN: u8 = 24_u8;
    const PREAMBLE_LEN: u16 = 12;

    const RF_FREQ: RfFreq = RfFreq::F433;

    const LORA_PACKET_PARAMS: LoRaPacketParams = LoRaPacketParams::new()
        .set_crc_en(true)
        .set_preamble_len(PREAMBLE_LEN)
        .set_payload_len(DATA_LEN)
        .set_invert_iq(false)
        .set_header_type(HeaderType::Fixed);

    const LORA_MOD_PARAMS: LoRaModParams = LoRaModParams::new()
        .set_bw(LoRaBandwidth::Bw125)
        .set_cr(CodingRate::Cr45)
        .set_ldro_en(false)
        .set_sf(SpreadingFactor::Sf7);

    const PA_CONFIG: PaConfig = PaConfig::LP_14;

    const TX_PARAMS: TxParams = TxParams::LP_14.set_ramp_time(RampTime::Micros40);

    let mut radio = SubGhz::new(dp.SPI3, &mut dp.RCC);

    radio.set_smps_clock_det_en(true).unwrap();
    radio.set_smps_drv(hal::subghz::SmpsDrv::Milli60).unwrap();

    radio.set_standby(hal::subghz::StandbyClk::Rc).unwrap();

    radio.set_hse_in_trim(HseTrim::from_raw(0x20)).unwrap();
    radio.set_hse_out_trim(HseTrim::from_raw(0x20)).unwrap();

    radio
        .set_regulator_mode(hal::subghz::RegMode::Smps)
        .unwrap(); // Use DCDC

    radio
        .set_buffer_base_address(TX_BUF_OFFSET, RX_BUF_OFFSET)
        .unwrap();

    radio.set_pa_config(&PA_CONFIG).unwrap();
    radio.set_pa_ocp(hal::subghz::Ocp::Max60m).unwrap();
    radio.set_tx_params(&TX_PARAMS).unwrap();

    radio
        .set_packet_type(hal::subghz::PacketType::LoRa)
        .unwrap();
    radio
        .set_lora_sync_word(hal::subghz::LoRaSyncWord::Public)
        .unwrap();
    radio.set_lora_mod_params(&LORA_MOD_PARAMS).unwrap();
    radio.set_lora_packet_params(&LORA_PACKET_PARAMS).unwrap();
    radio.calibrate_image(CalibrateImage::ISM_430_440).unwrap();
    radio.set_rf_frequency(&RF_FREQ).unwrap();

    // unsafe {
    //     subghz::unmask_irq();
    // }
    // Set IRQs
    radio
        .set_irq_cfg(
            &CfgIrq::new()
                .irq_enable_all(Irq::RxDone)
                .irq_enable_all(Irq::TxDone)
                .irq_enable_all(Irq::Timeout)
                .irq_enable_all(Irq::Err),
        )
        .unwrap();
```

And transmit dummy data code:
```
    let mut payload: [u8; 24] = [0u8; 24];
    for i in 0..24 {
        payload[i] = i as u8;
    }

    defmt::info!("Send data");
    radio.write_buffer(TX_BUF_OFFSET, &payload[..]).unwrap();
    radio_switch.set_tx();  // Turn on ctrl1, ctrl2 and ctrl3
    radio.set_tx(Timeout::from_millis_sat(500)).unwrap();

    loop {
        let (status, irq_status) = radio.irq_status().unwrap();
        let mode = status.mode().unwrap();
        let cmd = status.cmd().unwrap();
        radio.clear_irq_status(irq_status).unwrap();
        if irq_status & Irq::TxDone.mask() != 0 {
            // nothing is required upon TX done
            // you may want to put the radio into RX mode after TX completion
            defmt::info!("TX done");
            break;
        } else if irq_status != 0 {
            defmt::info!("Other IRQ")
            // ... handle other IRQs
        }
    }
```

I am using Msi 48Mhz clock.

Any hints are much appreciated and after figuring it out I plan to share my solution with the others in a form of an example code.

newAM, 2023-10-29:
The `subghz` test has a basic LoRa example: <https://github.com/stm32-rs/stm32wlxx-hal/blob/0d169de62c6801c744d1f9de522474c8da27ec93/testsuite/src/subghz.rs#L424>

You code looks correct.  `ExecutionFailure`s can be difficult to debug. Generally I start with a known working configuration then bisect to find why the radio is returning `ExecutionFailure`.

Perhaps try changing the clock/power configuration to match the code in the test?  That's one difference I noticed.

kolegs, 2023-10-30:
Examples helped a lot, not sure what was the issue but I used the code from the test with small changes and at least I am not getting `ExecutionFailure`. And I am receiving a lot of data (I think there is a plenty of smart meters in my area). Still cannot receive `PING` message, but I think it is a matter of time to make it work or find emptish frequency.

kolegs, 2023-10-31:
@newAM I am trying to make it work but for some reason I am not getting the correct data. Looks like I am sending the data correctly, I am getting the `TxDone` interrupt as expected.
But the reading part is a problem, I am not getting any data and the length is kind of faulty also I am getting `Irq::Err` which means the CRC is incorrect, but the buffer is empty (no data change at all, I check it before and after the read part).
I am pretty confident that I am triggering the interrupt from the other board, I've configured to send a data on button push and after every push I am entering the interrupt.

Any idea what might be cause? Is it possible that something else is also sending data at exactly same moment? I find it strange that I get exactly the same `Err` when I am using only one board, I get it just after transmitting the data.

newAM, 2023-11-01:
> Is it possible that something else is also sending data at exactly same moment?

It's possible, but unlikely I think.

---

I have not seen this sort of behavior before, so all the advice I can give is guesswork.  Are the radios configured the same in terms of hardware and code?

kolegs, 2023-11-07:
Anyway looks like something is wrong with my code, I have tested embassy example and it works like a charm there. I get the correct length and data. 
@newAM thanks for help and if there is anything I can help you with this repository let me know, I am eager to do it :smile:

closed 2023-11-07 by kolegs